Terrapin-Rocket-Team/Payload #1: MULE Flight updates

nate-roy, 2024-09-04:
- [ ] MULE PCB Validation
> - [ ] Confirm nichrome firing
- [ ] MULE Deployment mechanism validation
> - [ ] Waterjet new top section plates
> - [ ] Assemble and test deployment mechanism
- [ ] Mule sealing validation
> - [ ] Create new nose cone sled

nate-roy, 2024-09-25:
Cut wind tunnel plates to size for water jetting

abharm, 2024-10-30:
I added holes to the side panels of the MULE and minimized the bottom ballast, reducing total weight from 9.1 lbs to 6.7 lbs. Also did FEA for one of the side panels for a static force of 300 lbs, resulting in around 12 SF, meaning more weight can be reduced. 

[image]

[image]

abharm, 2024-11-06:
11/6

Cut larger holes in sides for weight
Added crossbar to hold prop arm in
took out hidden side panels
increased height to 40cm
made ballast block into quarter inch plate
Made 3 screws on each side of ballast plate
Cut hole in top plate (?)
Redid FEA - Min SF of 3.2 now

[image]

[image]

abharm, 2024-11-13:
Cut out some more holds for weight reduction
currently at ~5.3lbs

abharm, 2024-11-20:
11/20

lowered ebay, larger weight saving holes

[image]