Terrapin-Rocket-Team/Payload #2: MULE Serial connection

nate-roy, 2024-09-04:
- [ ] Teensy-Teensy serial communication
- [ ] Teensy-Teensy serial disconnect robustness
- [ ] Remote programing

aaronjj925, 2024-10-23:
Created sample code for communication between the two teensies; haven't tested, so will try to go to Cypress and test it out early next week

aaronjj925, 2024-10-28:
Uploaded the code to each teensy separately, but when I wired them together, they "merge" their code and prints out whatever the teensy that was directly plugged into the computer is supposed to print

aaronjj925, 2024-11-11:
Teensy 1: initial sender
Teensy 2: initial receiver

Teensy 2 is able to read data from teensy 1 and print the message, but teensy 1 is currently not receiving the message back from teensy 2 after first sending it

aaronjj925, 2024-11-20:
-couldn't work on it much, but did fix message being sent multiple times
-message receiving teensy is still not sending back

aaronjj925, 2024-12-04:
-couldn't work on it much again
-still not sending back and forth