Terrapin-Rocket-Team/Payload #5: Tail Rotor Development

nate-roy, 2024-09-04:
- [ ] H bridge hardware/software driver validation
> - [ ] Determine max frequency
- [ ] Motor characterization
- [ ] Motor simulation
- [ ] Motor validation
- [ ] Controller validation
- [ ] Chute tilt activation design
- [ ] Locomotion validation

rnorthcu19, 2024-09-18:
I will be going to cypress later in the week to run a few tests on the part and print a new version with increased infill/wall thickness.

osapa7, 2024-09-18:
Started control simulation

nate-roy, 2024-09-18:
Micro on test board is unable to output 5v from signal unless the usb cable is connected (dont know why, also struggling to deal with pull up resistors), and so cant trigger the h-bridge. Will try updated pull up resistors before switching to teensy testing board.
Print is complete, will be assembled soon

nate-roy, 2024-09-24:
Moved test board over to a teensy, integrated motor, battery, and board into test hardware. System is ready for testing, but might need a new 9v.

[image]

rnorthcu19, 2024-09-25:
Printed new arm, looked over nichrome stuff, and am still a bit confused but I can sort it out I think.

osapa7, 2024-09-25:
Making progress on the simulator. Want to ask some questions and get some dimensions of the payload.

rnorthcu19, 2024-10-19:
I created an initial model of a nichrome arm for the release system.

rnorthcu19, 2024-10-23:
I have modeled a second potential design. The next steps are printing and dropping-testing these designs to find possible flaws.

CooperBrownUMD, 2024-10-23:
Updated testing sled for LIPOs
[image]

rnorthcu19, 2024-10-30:
Adjusted the nichrome arms to fit the fishing line

rnorthcu19, 2024-10-30:
[image]

nate-roy, 2024-10-30:
Verified analogWriteFrequency() does what it says it does, so can switch to using that.

CooperBrownUMD, 2024-11-06:
Experimentally derived cubic fit mapping angular accelerations to PWM values.
[image]

osapa7, 2024-11-13:
Used derived angular acceleration to PWM relations to simulate angular position and velocity at given PWM.

rnorthcu19, 2024-11-13:
Printed out arms and mount, will need to redesign

osapa7, 2024-11-20:
Working on updating the simulator to make descrete calculations and updates with position and velocity calculations

osapa7, 2025-01-16:
Uploaded sim to git. Updating control algorithm for better tracking