Terrapin-Rocket-Team/Payload #6: Live video development

nate-roy, 2024-09-04:
- [ ] Gimbal structure design
- [ ] Camera orientation simulation
- [ ] Tadpol-avionics interface
- [ ] Live video transmission
- [ ] Camera orientation physical test
- [ ] RF interference verification

nate-roy, 2024-09-18:
Decided on converting payload-rocket vector to body frame via quaternion DCM, then computing angles. Didnt create a full simulation considering the implementations not that bad; its not worthwhile figuring out error for a body~=inertial system or euler approximation. Next step is L1 gimbal design

abharm, 2025-02-10:
2/10

Made simple camera gimbal, can improve if needed 
- also need to actually attach to vehicle

[image]

Also wrote code to find angles for camera to point from 
- approximated lat and long as flat coordinate plane
- resulted in small error of angle, I think is fine for small distances
- need to figure out integration with existing code to get gps and barometer data of vehicle and rocket