Terrapin-Rocket-Team/Payload #7: Navigation development

nate-roy, 2024-09-04:
- [ ] Performance simulation
- [ ] Software-limited turn rate validation
- [ ] Locomotion validation
- [ ] Wind correction validation
- [ ] Landing point orbit validation

mallamacimj, 2024-09-11:
# 9/11/24

## Did this week
- Nothing lol

## Next Week
- Euler Validation
- Look over the wind correction code from last year

ChenJeff123, 2024-09-11:
Didn't get anything done.
Next week will look over old descent path code and integrate wind correction code

rowleyhc, 2024-10-28:
10/28: Finished writing and testing some basic navigation simulation that incorporates the angle limitation command: 
[image]
[image]
[image]
From playing around with different targets and starting headings/points, it seems like the goalCommand function(which effectively limits the difference between the goal and actual angle that the controller uses) is working properly, and doesn't cause any issues. However, as can be seen the the heading is still changing very rapidly(about 180 degrees in 0.1 seconds, which seems to be the maximum the controller will output(1800 degrees/second)), which is still going to cause some issues. A rough band-aid solution for now seems to be just multiplying pwm by 1/20, which produces a much more reasonable turn rate:
[image]
[image]
[image]
To do: figure out a better way to limit the turn rate so that the pwm output still allows the motor to be more responsive and isn't unnecessarily scaled down