Terrapin-Rocket-Team/Payload #40: Path Planning/Navigation

abharm, 2025-09-30:
Robust navigation code using Astra and making assumptions about how the rover will act. May be helpful to create a simulation in python first and then move to cpp once we start integrating everything.

Requirements:
Decision tree for navigation after landing has been detected
Waypoint generation
Path planning between waypoints

PazEdu, 2025-10-06:
Researched general navigation/path planning: what it means, requirements, common planning algorithms, and general implementation. 

so far boiled down my understanding to : perception, localization, planning, and control. 

Made a high level navigation loop/steps:
1. Sense environment -> build/refresh occupancy grid.
2. Localize -> get position (x, y, θ).
3. Plan -> find path to next waypoint.
4. Control -> compute motor commands (v, ω).
5. Repeat at TBD frequency. 

Going forward:
- May use IR-SIM just for python simulation because of its minimal hardware requirements.
- Further refine plan.
- get more familiar with Astra.
   -  continue coding w/ Astra in mind.

misha-berry, 2025-11-03:
Started python code with general navigation loop, uploaded [here](https://github.com/misha-berry/misha.github/blob/dd83aaf36e456480c6c408eed809d4cd9ad627c2/nav/nav.py). Continued to familiarize myself with Astra Code

closed 2026-09-09 by abharm